translationalneuromodeling/tapas #247: PhysIO - Siemens XA30

iidree, 2023-09-19:
Hi Lars,

Thanks for this toolbox - I've used it previously with data from a Phillips scanner, without any issues. Now, I would like to use the toolbox with data from a Siemens XA30 scanner which I'm having a little trouble with.

To provide some background, we have both PPU and ECG (4-channel) data, from multiband sequences. The physiological files are embedded within the DICOMs. We have exported the data from the scanner and used the readCMRRPhysio.m script to extract the physio data from the DICOMs. In this case, should the .log file obtained from this script be entered in the PhysIO toolbox, or can the physio data be read in another way. Also, is there any specifics we should be aware of when using the toolbox using data from the XA30 and in a non-standard format?

Many thanks and best wishes,
Iman

mrikasper, 2023-09-19:
Dear Imam,

Thank you for using the PhysIO Toolbox, happy to hear it worked well on Philips data!

In principle, you should be able to use the `.log` data from the [`readCMRRPhysio.m`](https://github.com/CMRR-C2P/MB/blob/master/readCMRRPhysio.m) function directly, see [our wiki page](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/MANUAL_PART_READIN#cmrr-sequence-wip-advanced-physio-logging). 

You would have to select `Siemens_Tics` as the format for `log_files.vendor` and then it should work, but don't forget to put the `*_Info.log` files as `log_files.scan_timing` as well.

Please let me know if that helps.

All the best,
Lars

peter-mri, 2023-09-22:
Hope not to hijack the original topic, but my question seems fitting as it is related to using Siemens XA30 platform, but with manual physio recording using Siemens interface. This yields .resp, .puls etc text files which are normally readily processed by PhysIO. However I noticed the .resp file now contains a bunch of repeated high-value numbers, excerpt is below. Has anyone else encountered this and would you know how to extract actual resp data from this?

`1 8 20 2 5002 LOGVERSION_RESP   3 6002 5002 RespChannelSelected: 0 6002 5002 MSGTYPE 103 6002 5002 currentCushionGain: 38 6002 5002 MSGTYPE 400 6002 1031 33554432 98304 67108864 163840 1022 33554432 98304 67108864 163840 1012 33554432 98304 67108864 163840 1012 33554432 98304 67108864 163840 1003 33554432 98304 67108864 163840 993 33554432 98304 67108864`