translationalneuromodeling/tapas #260: 2-level HGF implementation

a-yur, 2024-02-13:
Is an implementation of a 2-level version of the binary HGF available in TAPAS? From what I understand, the existing implementation of the `tapas_hgf_binary` in TAPAS requires to have at least three levels. 

We plan to model a stimulus sequence with a 2-level HGF and then use the trial-by-trial estimates as parametric modulators, similar to Stefanics et al. (2019, _NeuroImage_).

chmathys, 2024-02-13:
You can get a 2-level binary HGF by fixing the parameter kappa_2 to zero (i.e., to -Inf in logit-space). That cuts off the third level from any information, effectively removing it.

You should also fix the third-level parameters mu_3^(0) and omega_3 to (arbitrary) values in order to prevent the optimization algorithm from trying to estimate them.

kwellstein, 2024-11-09:
Hi everyone,

we have a smilar issue.

We are trying to fit a 2-level eHGF binary (toolbox v7.1) but keep running into an "unstable region for the startpoint". 
I tried multiple prior settings in the config file but cannot get anywhere. Maybe I am missing something fundamental though... Below the config file:

% Model name
c.model = 'ehgf_binary';

% Number of levels (minimum: 3)
c.n_levels = 2;

% Input intervals
% If input intervals are irregular, the last column of the input
% matrix u has to contain the interval between inputs k-1 and k
% in the k-th row, and this flag has to be set to true
c.irregular_intervals = false;

% Sufficient statistics of Gaussian parameter priors

% Initial mus and sigmas
% Format: row vectors of length n_levels
% For all but the first two levels, this is usually best
% kept fixed to 1 (determines origin on x_i-scale). The 
% first level is NaN because it is determined by the second,
% and the second implies neutrality between outcomes when it
% is centered at 0.
c.mu_0mu = [NaN, 1]; % tried 0 too
c.mu_0sa = [NaN, 0];

c.logsa_0mu = [NaN,   log(0.1)]; % tried log(1) too
c.logsa_0sa = [NaN,        0];

% Rhos
% Format: row vector of length n_levels.
% Undefined (therefore NaN) at the first level.
% Fix this to zero to turn off drift.
c.rhomu = [NaN, 0];
c.rhosa = [NaN, 0];

% Kappas
% Format: row vector of length n_levels-1.
% Fixing log(kappa1) to log(1) leads to the original HGF model.
% Higher log(kappas) should be fixed (preferably to log(1)) if the
% observation model does not use mu_i+1 (kappa then determines the
% scaling of x_i+1).
c.logkamu = [log(1)]; % also tried -Inf here
c.logkasa = [0];

% Omegas
% Format: row vector of length n_levels.
% Undefined (therefore NaN) at the first level.
c.ommu = [NaN,  -3]; % also tried -6, -8, -10
c.omsa = [NaN,   4]; % also tried 4^2

thank you for your help :)

kwellstein, 2024-11-12:
update:
We shouldnt have deleted the 3rd (or in some cases) 2nd column like the comments suggested. But still, the issue remains. This is the current config, I tried the same values as above.
The 3-level HGF fits without an issue. I also tried fitting the 2-level Hgf using random inputs, the error persists. The task has 280 trials.

c.mu_0mu = [NaN, 0, 1];
c.mu_0sa = [NaN, 0, 0];

c.logsa_0mu = [NaN,   log(0.1), log(1)];
c.logsa_0sa = [NaN,          0,      0];

% Rhos
% Format: row vector of length n_levels.
% Undefined (therefore NaN) at the first level.
% Fix this to zero to turn off drift.
c.rhomu = [NaN, 0, 0];
c.rhosa = [NaN, 0, 0];

% Kappas
% Format: row vector of length n_levels-1.
% Fixing log(kappa1) to log(1) leads to the original HGF model.
% Higher log(kappas) should be fixed (preferably to log(1)) if the
% observation model does not use mu_i+1 (kappa then determines the
% scaling of x_i+1).
c.logkamu = [log(1), log(1)];
c.logkasa = [     0,      0];

% Omegas
% Format: row vector of length n_levels.
% Undefined (therefore NaN) at the first level.
c.ommu = [NaN,  -3,   2];
c.omsa = [NaN,   4,   0];

qiyunyi1230, 2024-12-19:
Hi everyone, I have also encountered this error When I tried to apply the two-level HGF model.: "Model inversion aborted. No stable startpoint found for the current priors in 1000 startpoint sampling iterations." Could you please let me know if there is a solution to this issue? Thank you very much!

chmathys, 2024-12-22:
Hi - you could use simulation to explore the range of parameter settings that produce realistic trajectories and then choose your priors to cover that range, but not more. These priors will include many stable start points for the optimization algorithm.

qiyunyi1230, 2024-12-23:
Dear Professor Mathys,

Thank you very much for your reply! I will definitely give this method a try later. Thanks to Dr. Schlossmacher's help, I no longer encounter this error. She advised me not to set c.nlevels to 2.

I have another small question: when fixing ka2 to 0, should I also fix its variance to 0? Considering that it is in log-space, should I then set logkasa2 = -Inf? 

Thank you again for your guidance!

chmathys, 2024-12-23:
Yes. Unless the prior variance is zero, the parameter is not fixed.

Good to hear you’ve fixed the starting point issue.

Best wishes,
Chris

qiyunyi1230, 2024-12-25:
Dear Professor Mathys,

I have a further question regarding the optimization procedure in the 'tapas_fitModel.m' script, specifically within the 'optim' function. The relevant code snippet is:

%matlab
opt_idx = [r.c_prc.priorsas, r.c_obs.priorsas];
opt_idx(isnan(opt_idx)) = 0;
opt_idx = find(opt_idx);

This identifies parameters to be optimized by excluding those with prior variances (priorsas) set to 0 or NaN. However, if 'logkasa(2)' is defined in the logarithmic space and its variance is set to '-Inf' (to fix the parameter), wouldn't this instead include the parameter in the optimization process rather than excluding it? Thus, wouldn't the correct setting to fix 'logkasa(2)' be '0' instead of '-Inf'?

Additionally, when I set logkasa(2) = -Inf, I noticed that the log model evidence (LME) becomes NaN - 1.570796i. This behavior raises concerns about whether -Inf is the appropriate value for fixing the parameter.

From the default parameter configuration in 'tapas_ehgf_binary_config()', only om2 and om3 have non-zero and non-NaN priors for their variances. This indicates that, by default, the model only optimizes om2 and om3. Therefore, to reduce the three-level model to a two-level model, my understanding is that the necessary adjustments are:

%matlab
c_prc = tapas_ehgf_binary_config();
c_prc.logkamu(2) = -Inf; % Fix ka2 prior mean to 0
c_prc.omsa(3) = 0;       % Fix om3 prior variance to 0

However, I noticed that some papers do not set the prior variance of om3 to 0 when simplifying the model. I compared the results with and without setting the om3 variance to 0, and the difference in the modeled pwPE2 was on the order of e-14. This suggests that setting the prior variance of om3 to 0 may not be strictly necessary.

To summarize, in terms of implementation, is it correct to say that the only required step is fixing the prior mean of 'ka2' to 0, i.e.,

%matlab
c_prc.logkamu(2) = -Inf;

I would appreciate your insights on whether this simplification is valid or if there are other considerations I might be missing.

Best regards,  
Yun-Yi Qi

chmathys, 2024-12-25:
Dear Yun-Yi,

You are right, the prior variance should be set to 0, not -Inf. This is because the estimation takes place in log-space, i.e., we estimate log(kappa2), not kappa2 (of course, when the prior mean is set to 0, no estimation at all takes place). So in sum, both the mean and variance of the kappa2 prior should be 0, and with estimation in log-space that means that the mean of the prior of log(kappa2) should be -Inf and the variance of that prior should be 0. I'm sorry for any confusion.

Your observation regarding the prior of om3 is interesting. As you say, strictly speaking, it does not make sense to estimate it and therefore its prior variance should be set to 0. However, when it is cut off from any information by setting kappa2 to zero, then any estimation will necessarily lead to a posterior that is identical to the prior. In that case, LME is unaffected, as you've noticed. This is where LME, approximated as the negative variational free energy, truly shines as a model score: you are free to include any number of parameters in your model without hurting your complexity score as long as the data make no difference to your belief about them - in other words, as long as the posterior equals the prior. You are only penalized for complexity when the data lead to a change in a parameter's distribution. This reflects the fact that complexity is not a property of the model, but a property of the data, given a certain model. Criteria like AIC and BIC do not take sufficient account of this and penalize you for complexity simply for adding parameters, which can seriously distort model selection.

Best,
Chris