translationalneuromodeling/tapas #264: Error running PhysIO: Item matlabbatch: No repeat named physio

LLouedKhen, 2024-05-02:
Hi, 
I would like to create physiological regressors using a script but it keep crashing with the following message:
`error: Item matlabbatch: No repeat named physio`

The same batch works fine in the GUI, although I have to manually add the TAPAS toolbox to the gui Batch. TAPAS doesn't otherwise appear in the options under tools in the GUI. 
I have used TAPAS PhysIO before in preprocessing pipeline but I don't recall running into this problem. 

I am running scripts on SPM12 (updated), Matlab 2023a. 

Here is my code:

```
for i = 1:length(subjects)
    thisPhysPath = physPaths{i};
    cd(thisPhysPath)
    physFiles = dir('Sess*');
    physFile1 = physFiles(1).name;
    physFile2 = physFiles(2).name;
    thisMRPath1 = imgPaths1{i};
    thisMRPath2 = imgPaths2{i};
    thisNScan1 = imgNum1(i);
    thisNScan2 = imgNum2(i);
    for r = 1:2
        if r == 1
            physFile = physFile1;
            epifolder = thisMRPath1;
            nscans = thisNScan1;

        elseif r == 2
            physFile = physFile2;
            epifolder = thisMRPath2;
            nscans = thisNScan2;
        end
        spm('Defaults','fMRI');
        spm_jobman('initcfg');
        matlabbatch{1}.physio.log_files.vendor = 'Biopac_Mat';
        matlabbatch{1}.physio.log_files.cardiac = physFile;
        matlabbatch{1}.physio.log_files.respiration = physFile;
        matlabbatch{1}.physio.log_files.scan_timing = {''};
        matlabbatch{1}.physio.log_files.sampling_interval = [];
        matlabbatch{1}.physio.log_files.relative_start_acquisition = 0;
        matlabbatch{1}.physio.log_files.align_scan = 'last';
        matlabbatch{1}.physio.scan_timing.sqpar.Nslices = 40;
        matlabbatch{1}.physio.scan_timing.sqpar.NslicesPerBeat = [];
        matlabbatch{1}.physio.scan_timing.sqpar.TR = 2.48;
        matlabbatch{1}.physio.scan_timing.sqpar.Ndummies = 0;
        matlabbatch{1}.physio.scan_timing.sqpar.Nscans = nscans;
        matlabbatch{1}.physio.scan_timing.sqpar.onset_slice = 20;
        matlabbatch{1}.physio.scan_timing.sqpar.time_slice_to_slice = [];
        matlabbatch{1}.physio.scan_timing.sqpar.Nprep = [];
        matlabbatch{1}.physio.scan_timing.sync.nominal = struct([]);
        matlabbatch{1}.physio.preproc.cardiac.modality = 'ECG';
        matlabbatch{1}.physio.preproc.cardiac.filter.no = struct([]);
        matlabbatch{1}.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;
        matlabbatch{1}.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
        matlabbatch{1}.physio.preproc.cardiac.initial_cpulse_select.auto_matched.max_heart_rate_bpm = 90;
        matlabbatch{1}.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
        matlabbatch{1}.physio.preproc.respiratory.filter.passband = [0.01 2];
        matlabbatch{1}.physio.preproc.respiratory.despike = false;
        matlabbatch{1}.physio.model.output_multiple_regressors = 'multiple_regressors.txt';
        matlabbatch{1}.physio.model.output_physio = ['physio', num2str(r), '.mat'];
        matlabbatch{1}.physio.model.orthogonalise = 'none';
        matlabbatch{1}.physio.model.censor_unreliable_recording_intervals = false;
        matlabbatch{1}.physio.model.retroicor.yes.order.c = 3;
        matlabbatch{1}.physio.model.retroicor.yes.order.r = 4;
        matlabbatch{1}.physio.model.retroicor.yes.order.cr = 1;
        matlabbatch{1}.physio.model.rvt.no = struct([]);
        matlabbatch{1}.physio.model.hrv.no = struct([]);
        matlabbatch{1}.physio.model.noise_rois.no = struct([]);
        matlabbatch{1}.physio.model.movement.no = struct([]);
        matlabbatch{1}.physio.model.other.no = struct([]);
        matlabbatch{1}.physio.verbose.level = 2;
        matlabbatch{1}.physio.verbose.fig_output_file = '';
        matlabbatch{1}.physio.verbose.use_tabs = false;
        save('matlabbatch');
       
        spm_jobman('run',matlabbatch);
       

        copyfile('multiple_regressors.txt', 'epiFolder')
        copyfile('physio.mat', 'epiFolder')

        clear;clc

    end
end
```

Thank you so much for your help. 
Best, 
Leyla

mrikasper, 2024-05-02:
Dear Leyla,

Could you please try to replace `matlabbatch{1}.physio` by `matlabbatch{1}.spm.tools.physio` in all lines of your script?

All the best,
Lars

LLouedKhen, 2024-05-03:
Dear Lars,
Thank you for responding so quickly. Unfortunately, that didn't work. I
think the problem lies in the SPM integration. When I run
tapas_physio_init()
It exits with a successful install (see first image). However, I can't find
it in the SPM tools options on the GUIs. The PhysIO module runs only if I
add its config file to the batch GUI (see second image). And then when I
save that as a script, the lines are saved as:
matlabbatch{1}.physio
Thank you.
Best,
Leyla
[image: Screenshot 2024-05-03 at 13.48.24.png]
[image: Screenshot 2024-05-03 at 14.10.58.png]

LLouedKhen, 2024-05-03:
Hi again,
An update: I moved the file tapas_physio_cfg_matlabbatch.m up to the
tapas-master folder and it is now integrated into the spm toolbox options.
Thank you.
Best,
Leyla

mrikasper, 2024-05-04:
Dear Leyla,

I think this is a longstanding difficulty when installing PhysIO...it should automatically copy the  tapas/PhysIO/code folder to spm12/toolbox/PhysIO, but unfortunately, this doesn't always work... I guess as soon as you put  tapas_physio_cfg_matlabbatch.m in a direct subfolder of spm12/toolbox, this also did the trick.

All the best,
Lars

https://github.com/LLouedKhen/