translationalneuromodeling/tapas #273: How to determine certain PhysIO parameters for Philips scanphyslog

Jothi-Ni, 2024-07-19:
Dear Lars,

I am trying to use the PhysIO toolbox for my Philips scanphyslog files and I am not sure whether I entered the correct values for some of the required parameters: 

- **log_file.log_scan_timing**: Do I need to add the same scanphyslog.log file as I entered for log_cardiac and log_respiration?
- **log_files.relative_start_acquisition**: The Philips log files do contain a marker column, where the value 0010 indicates the start of the MRI scan. In order to calculate the relative_start_acquisition I determined the number of measurements at which the starter marker occured (e.g. in the 29900 th line) and I calculate 29900/sampling rate in order to get the start of the MRI acquisition relative to the start of the scanphyslog file in seconds. Is that the correct approach or does PhysIO automatically recognize the 0010 start marker in the log file?
- **log_files.align_scan**: The description says "align the last scan to the end of the logfile". However, in the case of Philips, there is again a dedicated stop marker in the log file (value of 0020 in the marker column) which usually appears few seconds before the actual physlog file ends. Wouldn't there be a small mismatch then or should I remove those last lines of physiological data points after the end of the MRI scan from my scanphyslog file in the first place? 
- **scan_timing.sqpar.onset.slice**: I don't know how to determine this value. My fMRI experiment with a block-design immediately starts with the task block, as soon as the scan starts, so I would set this value to 0? With this, also Ndummies would be 0, correct?
- **scan_timing.Scan/Physlog time synchronization.gradient_log**: How do I know which of x,y,z to set for grad_direction? Does this depend on the slice acquisition direction? Moreover, can you please tell me how to choose values for gradient_log.zero and gradient_log.slice? 

Some information on my own setup: 

- I am using a 3T scanner with the wireless physiological sensors (sampling rate=496Hz). My fMRI sequence has a TR=2.9s, Nslices=53, Nscans=152.

- This is an example, of how such a scanphyslog file looks like
[fmri_verbalfluencytask_350_scanphyslog.log](https://github.com/user-attachments/files/16311170/fmri_verbalfluencytask_350_scanphyslog.log)

Thank you,

Jothini

mrikasper, 2024-08-27:
Dear Jothini,

Thank you so much for the detailed description of your question and for trying out the PhysIO Toolbox. Let me try to anwer your questions inline below

> 
> * **log_file.log_scan_timing**: Do I need to add the same scanphyslog.log file as I entered for log_cardiac and log_respiration?

Yes. For Philips, this information is all in one file.

> * **log_files.relative_start_acquisition**: The Philips log files do contain a marker column, where the value 0010 indicates the start of the MRI scan. In order to calculate the relative_start_acquisition I determined the number of measurements at which the starter marker occured (e.g. in the 29967 th line) and I calculate 29967/sampling rate=29967/496Hz=60.42 s in order to get the start of the MRI acquisition relative to the start of the scanphyslog file in seconds. Is that the correct approach or does PhysIO automatically recognize the 0010 start marker in the log file?
>   Because later when PhysIO runs, it says "tmin (1st scan start (1st dummy))=  86.42 s" in the command window, even though I specified a start_aquisition of 60.42 seconds.

I haven't scanned on a Philips system in the last 5 years, but did so for more than 10 years before that. In all that time, I never got to understand what the start and stop marker really mean in the logfile. They never quite aligned with the start of the first imaging volume or end of the last one - maybe they have fixed that in the meantime, but if not, I would suggest that you just leave this field at 0 and let PhysIO try to figure out the scan synchronization from the gradient timecourses.

> * **log_files.align_scan**: The description says "align the last scan to the end of the logfile". However, in the case of Philips, there is again a dedicated stop marker in the log file (value of [...]

No, you don't have to edit your logfiles. It should work out of the box. The reason I recommend aligning to the last scan on Philips systems is that logging starts as soon as you hit the Start Scan button on the console, so the physiological logfiles contains all the preparations (shimming) and dummy scans including their gradient timecourses before the first volume is acquired. It's much easier to count from the end, where the logging ends shortly after the scan stops.

> * **scan_timing.sqpar.onset.slice**: I don't know how to determine this value. My fMRI experiment with a block-design immediately starts with the task block, as soon as the scan starts, so I would set this value to 0? With this, also Ndummies would be 0, correct?

`onset_slice` should be the index of the slice you want the regressors to be aligned with. Typically, I would set it to `nSlices/2` where `nSlices` is the total number of slices in the scan volume.

> * **scan_timing.Scan/Physlog time synchronization.gradient_log**: How do I know which of x,y,z to set for grad_direction? Does this depend on the slice acquisition direction? Moreover, can you please tell me how to choose values for gradient_log.zero and gradient_log.slice?

There is a short description in the [PhysIO FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#12-philips-i-would-like-to-use-the-gradient-log-for-timing-synchronization-but-how-do-i-set-the-thresholds). Let me know if you need more pointers.

> 
> Some information on my own setup:
> 
> * I am using a 3T scanner with the wireless physiological sensors (sampling rate=496Hz). My fMRI sequence has a TR=2.9s, Nslices=53, Nscans=152.
> 

This all makes sense.

> -This is the figure created, when I run PhysIO with my values: ![image](https://private-user-images.githubusercontent.com/91665949/350375168-82427e48-2d5b-4446-bc46-e9e438b2d23d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3OTA5NDEsIm5iZiI6MTcyNDc5MDY0MSwicGF0aCI6Ii85MTY2NTk0OS8zNTAzNzUxNjgtODI0MjdlNDgtMmQ1Yi00NDQ2LWJjNDYtZTllNDM4YjJkMjNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDIwMzA0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxNzk5NjgxMTdjZWVkMTU5NDk3Mzk4NTEyODE4Njc5MDU0ZmIzM2E1YjI2MGNjMzkyMzI3N2Y2YzE4NDMzZjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4SIcawxMv27LYF_otk1fvgcnJDt3eh5u5CT1sBjNEDI)
> 

If you zoom in a bit to see only a 20s window or so, you should be able to discern regularities with the volume or slice TR in at least one of the gradient time courses (top panel), and maybe regular gaps at the end of a volume TR (after the last slice) in the bottom one, which should help you select the thresholds for the slice or volume onset detection.

> * This is an example, of how such a scanphyslog file looks like
>   [fmri_verbalfluencytask_350_scanphyslog.log](https://github.com/user-attachments/files/16311170/fmri_verbalfluencytask_350_scanphyslog.log)

Nothing unexpected here from first glance. I think you should just go ahead and ajdust the thresholds for the gradient peak detection until you get a reasonable result. You only have to do this once for your study and can leave them the same for your other subjects (unless you change scan geometry a lot between participants).

I hope that helps. Let me know if you get stuck with the thresholding.

All the best,
Lars

Jothi-Ni, 2024-10-04:
Dear Lars,

thanks a lot for your answers, this helped a lot! 

I still have a few questions left for scan/physlog time synchronization after reading the FAQ section:

1. Based on the explanation in the FAQ I selected gradient_log.grad_direction to z, as gradient z is showing the highest peaks in my case (see below). Is that correct? I chose the following values for gradient_log.zero=0.48 and gradient_log.slice=0.54 based on the recommendation from the FAQ. I left gradient_log.vol and gradient_log.vol_spacing empty.

This is how the raw gradient time courses look like when I zoom in. 
[image]
Zooming in even more:
[image]

2. Regarding your other comment on relative_start_acquisition: 
> I would suggest that you just leave this field at 0 and let PhysIO try to figure out the scan synchronization from the gradient timecourses.

So you mean setting it to 0 (and not empty) which would mean simultaneous start of physlog file and scan start, which is not the case for my study as the physlog seems to start some seconds before the actual scan starts. If I leave this field empty, however, I run into an error message (Failed  'TAPAS PhysIO Toolbox'
Error using  +   Matrix dimensions must agree.). If I set it to 0, I at least don't get an error message and PhysIO runs without errors. Does the scan synchronization from the gradient time courses in this case overwrite/ignore the relative_start acquisition time? 

3. Can the number of found volume events be seen  as a check to verify whether PhysIO runs correctly on my dataset? In my case it found 152 volumes and that is indeed the number of scan volumes which I specified for this fMRI sequence. 

4. Eventually the generated multiple_regressors.txt file can then be used as an additional regressor in my first level model in SPM for example, correct?

Thank you,

Jothini

Jothi-Ni, 2024-11-06:
Dear @mrikasper ,

Do you have any thoughts on the questions I posted previously (Oct 4)?
Your help is very much appreciated and would help me to proceed with my analysis.

Thank you very much.

Jothini