translationalneuromodeling/tapas #279: Using Physio to obtain physiological metrics without MRI noise correction

tantantree, 2024-08-21:
Dear Lars,

I am trying to use PhysIO to analyze my physio trace (breathing belt and pulse oximeter) collected from a Siemens Prisma 3T. I have noticed that PhysIO requires me to input MRI-related data (such as Nscans; relative_start_acquisition, etc.), but I am **only interested in analyzing the physiological signals to obtain the average heart rate/pulse rate per minute/ HRV/ RVT during the scans**. Is it possible to achieve this without providing MRI-related parameters?

I also tried running PhysIO with MRI data, but encountered an error: "Index exceeds the number of array elements. Index must not exceed 0."
In file "/Users/guanqi/Downloads/tapas-master/PhysIO/code/preproc/tapas_physio_simulate_pulse_samples.m" (???), function "tapas_physio_simulate_pulse_samples" at line 102.
In file "/Users/guanqi/Downloads/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles.m" (???), function "tapas_physio_read_physlogfiles" at line 107.
In file "/Users/guanqi/Downloads/tapas-master/PhysIO/code/tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 137.
In file "/Users/guanqi/Downloads/tapas-master/PhysIO/code/tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1661.
It seems that the calculated value of "meanPulseRateInSamples" is very abnormal, which is causing the failure to compute simulatedPulses. If I must use MRI data to calculate physiological metrics, could you help me troubleshoot this error?
[image]

Thank you so much!

Warm wishes,
Anqi

mrikasper, 2024-08-27:
Dear Anqi,

PhysIO uses the sequence parameter information to estimate how long the time series should be that it focuses on (and later create regressors for the right number of volumes, but that is not important in your case). 

So, if you have a rough estimate in seconds how long the time series is that you want to analyze, you can just put `nScans = durationTimeSeries_seconds` and `TR =1`, and `nSlices =1` (maybe 2, if 1 gives an error).

If your estimate is longer than the actual time series in the files, then PhysIO tries to extrapolate data which could lead to the `simulatedPulses` you encounter and maybe relates to the error you see. Rather choose the duration as described above to be something a little bit shorter than the expected length of the physiological recordings.

I hope that helps.

All the best,
Lars

tantantree, 2024-09-02:
Dear Lars,

Thank you for your timely reply, it gives me the hope of solving this problem. I tried TR =1 and 2; nScans = 200 (which is definitely shorter than my physiological recordings duration). The physiological recordings duration was 4min 58s for pulse and 4 min 59s for respiration, calculated from .resp and .plus log file, respectively. 
[image]

However, I met the same error after adjusted the parameters.
[image]

It seems that PhysIO was still trying to simulate the data. Should I change the value of "relative_start_acquisition"?

Thank you for your patience and support!

Best,
Anqi

tantantree, 2025-01-07:
Dear Lars,

I still couldn't find a solution for this issue, could you please let me know if further actions I can try? It will help a lot and thank you for your precious time on developing this wonderful toolbox!

Best,
Anqi

mrikasper, 2025-01-07:
Dear Anqi,

My apologies for the delay and happy 2025! 

One thing I only spotted now is that you specified a sampling interval of 20 - however the unit is seconds, this would be unusually large. Try 0.02 if the interval was 20ms, or leave it empty, because PhysIO should be able to infer the sampling interval from the Siemens data itself.

If that does not help, would you be able to send me an example logfile together with the matlabbatch you have used, so that I can try to reproduce the error?

Thank you!

All the best,
Lars

tantantree, 2025-01-08:
Dear Lars,

Thank you so much for your timely reply! I changed the sampling interval to 0.02 then run and it generated 13 figures with no errors. It means the data was successfully processed, I think? I read your detailed elegant FAQ link for PhysIO (https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ). Question 9 is very close to my situation. However, I checked physio.ons_secs structure and found that my physio.ons_secs.rvt and physio.ons_secs.hr are zeros. Could you please tell me why this happened? I attached my batch file and the log file for your reference.

Happy New Year, and all best wishes for your wonderful and fruitful new year!
Anqi

[Anqi_physioTest.zip](https://github.com/user-attachments/files/18345652/Anqi_physioTest.zip)

mrikasper, 2025-01-08:
Dear Anqi,

I'm glad this helped.

To populate the hr and rvt field, you have to select the respective models to be computed (HRV and RVT), scroll a bit down in the batch editor view of your matlabbatch and set these two models to "yes", keeping the suggested default parameters.

All the best,
Lars

tantantree, 2025-01-08:
Dear Lars,

I think we are quite close to the success! I turned on the two models and the HR and RVT showed up! Please allow me to ask a silly question: the HR data I got is about 8, which means that the heart rate per scan is 8. Isn't this weird? Cuz I set the TR = 2, and it means that this heart beated 240 times per minute. Do I understand the calculation logic incorrectly or was the heart too hard-working...? Besides, the calculated HRV is about 0.6 to 0.9, is this normal?

Thanks and regards,
Anqi