translationalneuromodeling/tapas #280: FFT analysis problem with respiratory data

olympia83, 2024-08-27:
Hi! 
We are currently performing an FFT analysis on our fMRI data, with a particular focus on consciousness in participants exhibiting varying levels of arousal (sleep deprived and sedated) . We would like to  incorporate respiratory rate, volume, and depth into our analysis.
We have been utilizing the PhysIO toolbox in SPM, following the approach outlined by Kasper et al. While the documentation indicates that respiration should be included (with three regressors in addition to HR x CRF), we are not observing these components in our output graphs. We are encountering some challenges there.
Any advice or suggestions would be greatly appreciated.
Thank you very much for your time and assistance.

mrikasper, 2024-08-27:
Dear @olympia83,

Would you be able to provide some output plots from the toolbox, for example the "Diagnostics plot", please? 

Also, when you say that you didn't observe components in the output graphs, are you referring to the fact that the regressors were not created or that the F-contrasts including those regressors didn't not show any significant voxels (e.g., when using `tapas_physio_report_contrasts`)?

Thank you!

olympia83, 2024-08-28:
Thanks a lot for you answer. I am attaching the figure. Should we have to get three more regressors after HR x CRF right?
[figures_09](https://github.com/user-attachments/assets/821fd8a4-799e-4fc3-8be2-8929575d1b35)
Thanks 
/Olympia