translationalneuromodeling/tapas #281: Negative logF and zero Ep.B matrix in rDCM

ZiyiHuang-study, 2024-09-23:
Dear developers,

I am new to the rDCM in TAPAS, and  I had a question when using it on both my task-based and resting-state fMRI data. 

I ran "tapas_rdcm_model_specification" and "tapas_rdcm_estimate" in MATLAB, and no errors were reported, but the question was: output.logF was always a negative value, output.statistics were all empty, and output.Ep.B matrix was all zeros. 

I tried different numbers of brain regions (6 or 30), different subjects, the original or sparse methods, or predefining the args.a as the rsFC, but the question was still there. I tried to define DCM.b to be all ones without using the sparse method for my task data, which would let the output.Ep.B matrix be non-zeros (but very small <0.05), but the model showed negative logF and empty statistics. 

It seems that the model fitting was not good, but I don't know how to fix it. Would you mind making any comments on this issue? Waiting for your response!

Best regards
Ziyi

ImreKertesz, 2024-09-23:
Dear Ziyi,

output.logF represents the negative free energy (and not the log of the negative free energy as the name would suggest). Therefore, the value can be negative.
output.statistics is empty because you apply the model to empirical data. The field is none empty when you apply it to synthetic data where you have ground truth data generating parameters (output.statistics stores information on how well parameter recovery worked).
rDCM is not able to estimate the B matrix because it's based on a linear DCM (no modulatory influences). Therefore, the output.Ep.B matrix will always contain zeros.
This means all your observations are expected. Based on the provided information model inversion worked fine.
I hope this information is helpful.

Best regards,
Imre

ZiyiHuang-study, 2024-09-23:
Dear Imre,

Thanks a lot for your quick and detailed response! So, the rDCM toolbox may not be suitable for me to estimate the B matrix although it could be applied to task data? I would really appreciate it if you could tell me how to estimate the A and B matrix simultaneously (maybe other toolboxes) as long as you know (sorry it's out of the scope). 

Best regards,
Ziyi

ImreKertesz, 2024-09-24:
Dear Ziyi,

Yes, the rDCM toolbox is not suited if you are interested in the B matrix. This doesn't prohibit you from applying rDCM to task data since it has a driving input u which can encode information about your experimental paradigm.
If you are modeling only a few regions (max. 9-10) you can use the original formulation of DCM (implemented in SPM) to estimate the B matrix.
In the case where each session represents a certain condition (as in this [paper](https://www.sciencedirect.com/science/article/pii/S1053811920309769) with the hand movement paradigm) you can apply rDCM to each session separately and circumvent the problem of estimating the B matrix since no modulatory influences are required.

Best,
Imre

ZiyiHuang-study, 2024-09-25:
Dear Imre,
 
Thank you very much for your quick answer! My experiment is an event-related design, which may not be the case as you mentioned. I will close this issue.

Best,
Ziyi

closed 2024-09-25 by ZiyiHuang-study