translationalneuromodeling/tapas #285: Axes in tapas_rdcm_visualize connectivity matrices are mislabeled

hvannieuwenh, 2024-10-26:
Hi,

I believe the axes for the connectivity matrix figures (for example, 'estimated' and 'true') are mislabeled - the x axis is labeled as 'region (from)' and the y axis as 'region (to)', when it should be the other way around following the convention of figure 1 in the tutorial.

ImreKertesz, 2024-10-28:
Hi @hvannieuwenh ,

In the context of DCM the convention is to label the x axis (columns of the connectivity matrix) as "region (from)" and the y axis (rows of the connectivity matrix) as "region (to)". Both the function **tapas_rdcm_visualize** and figure 1 from the tutorial use this convention.
If you haven't changed anything in the code, figure 1 from the tutorial should label the axis using the same convention.

Btw. we recently released a more efficient Julia implementation of the toolbox. In case you're interested, you can find it here: https://github.com/ComputationalPsychiatry/RegressionDynamicCausalModeling.jl

Best,
Imre

hvannieuwenh, 2024-10-28:
Hi Imre,

Thanks for letting me know about the Julia implementation, I will definitely take a look. 

By that convention, node 1/region 1 in Figure 1 from the tutorial would not be connected to anything except for itself, since the entire column is empty, which does not make sense in the context of the S50 model. Fig 2. of the Smith 2011 paper also says "...where an element in the upper diagonal of the matrix implies a directed connection from a lower-numbered node to a higher-numbered one" and they have the same (albeit binarized) matrix. 

Hope I'm not misunderstanding something here - I just want to make sure I am interpreting my results correctly. 

Thanks,
Jolien

ImreKertesz, 2024-10-28:
Hi Jolien,

The first row of the A matrix has entries, indicating that there are connections going **to** region 1. As you wrote correctly, there are no entries in the column (besides the self-connection) of region 1, indicating that there no outgoing connections from region 1. This is in line with your quote from Smith 2011. In other words, region 1 is one of the few nodes that has only in-going but no out-going connections.
Essentially, A_{i,j} (an entry in the i-th row and j-th column) represents the connection from region j to region i.

I hope this helps.

Best,
Imre

hvannieuwenh, 2024-10-28:
Hi again,

I am still not sure that is correct - look at row 3 for example. It is the only row or column to have 5 connections, which is good because then symmetry will not confuse us here. By your convention there would then be 4 connections (ignoring self-connection) going **to** region 3, but there are no such regions in the network connectivity architecture figure. There is only one region in the bottom right of the top left cluster that has 4 connections going **from** it **to** other regions (the one with the long-range connection), and that would fit my convention.

Best,
Jolien

ImreKertesz, 2024-10-29:
Indeed, Fig 2 in Smith 2011 uses a different convention than in DCM. In that figure, an entry (i,j) denotes a connection from region i to region j (and not from j to i). This means the tutorial performs the analysis on a transposed version of the network presented in the paper. If you want to perform the analysis on the exact same network you need to transpose the respective matrices (defining the network architecture, priors, and ground truth parameters) after loading the DCM. Therefore, the difference comes not from the labeling of the plot but relates to how the network architecture is treated in this example.

Thank you for spotting that! We will clarify this difference in the tutorial script and documentation.

Best,
Imre