translationalneuromodeling/tapas #287: Siemens Physio (.phys/.puls)

HuwSwan, 2024-11-13:
Hello all,

I'm working with data from a Siemens 3T prisma, and having a bit of a synchronisation issue.

The log files are initiated manually from the terminal, thus both puls and resp files have different start times. In TAPAS I can only enter one value for relative_start_acquisition, meaning I have no way to align the physiology with the scans.

I tried createing a 'Custom' set of physiology data by just extracting the relevant physiology information during my acquisitions, saving them into txt files and running it manually, but this seems to also not work as the toolbox will hang for ~30 minutes on an EPI sequence about 1 minute in length. I assume this is because such manual alignment means there are no dummy scans or prep period, but I don't have any timing information for these, all I have is the initiation of the physio log, and the time of the first true volume acquisition.

What's the best way to navigate this? I haven't found a way to write ammended .resp/.puls files to manually align them, but I also am unfamiliar with any alternative way to resynchronise the data if the relative start times cannot be individually defined.

In the help text for 'relative_start_acquisition' there is mention of using an 'acquisition_info' file for synchronising things, which sounds like it would be an applicable thing to use, however I cannot find a single mention of what format this file should take or what it should containg in any documentation or help archives.

Thank you kindly.

HuwSwan, 2024-11-13:
Righto, I see.

This requires manually reading every DICOM to find the exactly relevant last volume from each acquisition if I understand correctly?

I had been hoping to avoid this as it's quite a slow process but can do!

mrikasper, 2024-11-13:
Dear @HuwSwan,

I don't think there is a lot of manual intervention necessary. You can just use the first volume in the DICOM folder of the respective run and select align_scan - 'first';

All the best,
Lars