translationalneuromodeling/tapas #295: 7T Siemens Terra - not a DICOM file

jskim628, 2025-01-13:
I get .IMA files from the 7T simens Terra. I created _INFO.log, _PULS.log, and _RESP.log files after using physio_convert.py. However, when I run the following script, I get this error.

Physio_20241114_125537_4e3a84e5-8091-4fe2-8e3f-34e68d6fd629_Info.log: Not
a DICOM file. 

Is a DICOM file required? If so, I only get the 1 .IMA file which I believe has all of the information. Should I just convert the .IMA to dicom and then put the same file for each of the logs? Thank you for your help.

```

clear all;
% Base directory for physio logs
base_dir = '/Users/andymac/Desktop/test';
mkdir physio_out/;
cd(base_dir);

% Loop through each subject and create the job script

subject = 'Physio_20241114_125537_4e3a84e5-8091-4fe2-8e3f-34e68d6fd629';
save_dir = fullfile(base_dir, 'physio_out');
cardiac_log = fullfile(base_dir, [subject '_PULS.log']);
respiration_log = fullfile(base_dir, [subject '_RESP.log']);
scan_timing_log = fullfile(base_dir, [subject '_Info.log']);

% Create matlabbatch structure
matlabbatch{1}.spm.tools.physio.save_dir = {save_dir};
matlabbatch{1}.spm.tools.physio.log_files.vendor = 'Siemens';
matlabbatch{1}.spm.tools.physio.log_files.cardiac = {cardiac_log};
matlabbatch{1}.spm.tools.physio.log_files.respiration = {respiration_log};
matlabbatch{1}.spm.tools.physio.log_files.scan_timing = {scan_timing_log};
matlabbatch{1}.spm.tools.physio.log_files.sampling_interval = [];
matlabbatch{1}.spm.tools.physio.log_files.relative_start_acquisition = 0;
matlabbatch{1}.spm.tools.physio.log_files.align_scan = 'last';
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nslices = 84;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.TR = 2;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Ndummies = 0;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nscans = 219;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.onset_slice = 50;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nprep = 0;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.scan_timing_log = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.modality = 'ECG';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.filter.no = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.max_heart_rate_bpm = 90;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.respiratory.filter.passband = [0.01 2];
matlabbatch{1}.spm.tools.physio.preproc.respiratory.despike = false;
matlabbatch{1}.spm.tools.physio.model.output_multiple_regressors = 'multiple_regressors.txt';
matlabbatch{1}.spm.tools.physio.model.output_physio = 'physio.mat';
matlabbatch{1}.spm.tools.physio.model.orthogonalise = 'none';
matlabbatch{1}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.c = 3;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.r = 4;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.cr = 1;
matlabbatch{1}.spm.tools.physio.model.rvt.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.hrv.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.noise_rois.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.movement.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.other.no = struct([]);
matlabbatch{1}.spm.tools.physio.verbose.level = 2;
matlabbatch{1}.spm.tools.physio.verbose.fig_output_file = 'physio.fig';
matlabbatch{1}.spm.tools.physio.verbose.use_tabs = false;

% Save the job script
save(fullfile(base_dir, 'physio_job.mat'), 'matlabbatch');

job_file = fullfile(base_dir, 'physio_job.mat');

% Load the job file into a variable
job_data = load(job_file);
matlabbatch = job_data.matlabbatch;

% Run the job
spm_jobman('run', matlabbatch);

% Close parallel pool
delete(gcp('nocreate'));

```

mrikasper, 2025-01-13:
Dear Andy,

No, you don't require a DICOM file for your physiological file format to synchronize with the scanner.
However, this looks like the new Siemens format. Can you set `log_files.vendor = 'Siemens_Tics';` and try again, please?

All the best,
Lars

jskim628, 2025-01-13:
Ah, that change fixed the issue. The toolbox ran successfully thank you so much.

closed 2025-01-13 by jskim628